geeksville/arduleader #180: java.lang.Throwable: A WebView method was called on thread 'Timer-0'. All WebView methods must be called on the same thread. (Expected Looper Looper (main, tid 1) {4229b070} called on null, FYI main Looper is Looper (main, tid 1) {4229b070})

geeksville, 2015-02-20:
```bash
java.lang.RuntimeException: java.lang.Throwable: A WebView method was called on thread 'Timer-0'. All WebView methods must be called on the same thread. (Expected Looper Looper (main, tid 1) {4229b070} called on null, FYI main Looper is Looper (main, tid 1) {4229b070})
at android.webkit.WebView.checkThread(WebView.java:2082)
at android.webkit.WebView.destroy(WebView.java:659)
at com.geeksville.gcsapi.WebActivity$$anon$1.run(WebActivity.scala:70)
at java.util.Timer$TimerImpl.run(Timer.java:284)
Caused by: java.lang.Throwable: A WebView method was called on thread 'Timer-0'. All WebView methods must be called on the same thread. (Expected Looper Looper (main, tid 1) {4229b070} called on null, FYI main Looper is Looper (main, tid 1) {4229b070})
at android.webkit.WebView.checkThread(WebView.java:2072)
... 3 more
java.lang.RuntimeException: java.lang.Throwable: A WebView method was called on thread 'Timer-0'. All WebView methods must be called on the same thread. (Expected Looper Looper (main, tid 1) {4229b070} called on null, FYI main Looper is Looper (main, tid 1) {4229b070})
at android.webkit.WebView.checkThread(WebView.java:2082)
at android.webkit.WebView.destroy(WebView.java:659)
at com.geeksville.gcsapi.WebActivity$$anon$1.run(WebActivity.scala:70)
at java.util.Timer$TimerImpl.run(Timer.java:284)
Caused by: java.lang.Throwable: A WebView method was called on thread 'Timer-0'. All WebView methods must be called on the same thread. (Expected Looper Looper (main, tid 1) {4229b070} called on null, FYI main Looper is Looper (main, tid 1) {4229b070})
at android.webkit.WebView.checkThread(WebView.java:2072)
... 3 more

```

ss56, 2015-05-01:
Hi I am facing similar issue, Please let me know if you had found any solution.

GggZX, 2016-07-06:
Maybe you should change the test wifi to another one and test it again

midhun-murali, 2017-02-20:
The used method is executed on a background (i.e. non-UI) thread. You need to call all Android View related methods on the UI thread. You can achieve what you need with:

final WebView mWebView = (WebView) findViewById(R.id.webView);
mWebView.post(new Runnable() {
    @Override
    public void run() {
        mWebView.loadUrl(...).
    }
});