geeksville/arduleader #192: libcore.io.GaiException: getaddrinfo failed: EAI_NODATA (No address associated with hostname)

geeksville, 2015-09-02:
```bash
java.net.UnknownHostException: Unable to resolve host "192.168.254.": No address associated with hostname
at java.net.InetAddress.lookupHostByName(InetAddress.java:424)
at java.net.InetAddress.getAllByNameImpl(InetAddress.java:236)
at java.net.InetAddress.getByName(InetAddress.java:289)
at com.geeksville.mavlink.MavlinkUDP.serverHost$lzycompute(MavlinkUDP.scala:29)
at com.geeksville.mavlink.MavlinkUDP.serverHost(MavlinkUDP.scala:29)
at com.geeksville.mavlink.MavlinkUDP.doSendMavlink(MavlinkUDP.scala:47)
at com.geeksville.mavlink.MavlinkSender$$anonfun$onReceive$1.applyOrElse(MavlinkSender.scala:41)
at akka.actor.Actor$class.aroundReceive(Actor.scala:465)
at com.geeksville.mavlink.MavlinkUDP.aroundReceive(MavlinkUDP.scala:24)
at akka.actor.ActorCell.receiveMessage(ActorCell.scala:516)
at akka.actor.ActorCell.invoke(ActorCell.scala:487)
at akka.dispatch.Mailbox.processMailbox(Mailbox.scala:241)
at akka.dispatch.Mailbox.run(Mailbox.scala:223)
at akka.dispatch.ForkJoinExecutorConfigurator$AkkaForkJoinTask.exec(AbstractDispatcher.scala:393)
at scala.concurrent.forkjoin.ForkJoinTask.doExec(ForkJoinTask.java:260)
at scala.concurrent.forkjoin.ForkJoinPool$WorkQueue.runTask(ForkJoinPool.java:1339)
at scala.concurrent.forkjoin.ForkJoinPool.runWorker(ForkJoinPool.java:1979)
at scala.concurrent.forkjoin.ForkJoinWorkerThread.run(ForkJoinWorkerThread.java:107)
Caused by: libcore.io.GaiException: getaddrinfo failed: EAI_NODATA (No address associated with hostname)
at libcore.io.Posix.getaddrinfo(Native Method)
at libcore.io.ForwardingOs.getaddrinfo(ForwardingOs.java:61)
at java.net.InetAddress.lookupHostByName(InetAddress.java:405)
... 17 more
java.net.UnknownHostException: Unable to resolve host "192.168.254.": No address associated with hostname
at java.net.InetAddress.lookupHostByName(InetAddress.java:424)
at java.net.InetAddress.getAllByNameImpl(InetAddress.java:236)
at java.net.InetAddress.getByName(InetAddress.java:289)
at com.geeksville.mavlink.MavlinkUDP.serverHost$lzycompute(MavlinkUDP.scala:29)
at com.geeksville.mavlink.MavlinkUDP.serverHost(MavlinkUDP.scala:29)
at com.geeksville.mavlink.MavlinkUDP.doSendMavlink(MavlinkUDP.scala:47)
at com.geeksville.mavlink.MavlinkSender$$anonfun$onReceive$1.applyOrElse(MavlinkSender.scala:41)
at akka.actor.Actor$class.aroundReceive(Actor.scala:465)
at com.geeksville.mavlink.MavlinkUDP.aroundReceive(MavlinkUDP.scala:24)
at akka.actor.ActorCell.receiveMessage(ActorCell.scala:516)
at akka.actor.ActorCell.invoke(ActorCell.scala:487)
at akka.dispatch.Mailbox.processMailbox(Mailbox.scala:241)
at akka.dispatch.Mailbox.run(Mailbox.scala:223)
at akka.dispatch.ForkJoinExecutorConfigurator$AkkaForkJoinTask.exec(AbstractDispatcher.scala:393)
at scala.concurrent.forkjoin.ForkJoinTask.doExec(ForkJoinTask.java:260)
at scala.concurrent.forkjoin.ForkJoinPool$WorkQueue.runTask(ForkJoinPool.java:1339)
at scala.concurrent.forkjoin.ForkJoinPool.runWorker(ForkJoinPool.java:1979)
at scala.concurrent.forkjoin.ForkJoinWorkerThread.run(ForkJoinWorkerThread.java:107)
Caused by: libcore.io.GaiException: getaddrinfo failed: EAI_NODATA (No address associated with hostname)
at libcore.io.Posix.getaddrinfo(Native Method)
at libcore.io.ForwardingOs.getaddrinfo(ForwardingOs.java:61)
at java.net.InetAddress.lookupHostByName(InetAddress.java:405)
... 17 more

```

rodrigAnd, 2020-10-30:
I have the same error..
Help, please!!

geeksville, 2020-11-11:
@rodrigAnd alas - it has been a long time since I updated this project so sorry, I think your on your own for this one (I haven't done drone things in the last few years)

closed 2020-11-11 by geeksville